AdaCore/Certyflie #1: Compilation problem : "Interfaces.Stm32.Rcc" is not a predefined library unit

VictorLSogilis, 2016-09-01:
Hi,

I'm trying to compile the ravenscar-cf-stable branch on commit a71ba5224097c885a6e5d6f8013b90a0d9af2d71

```
lambret@debian-sogilis:~/CAP2018/Certyflie$ gprbuild -P cf_ada_spark.gpr -p
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g main.adb
main.adb:30:06: "Interfaces.Stm32.Rcc" is not a predefined library unit
main.adb:30:06: "System.Bb.Parameters (spec)" depends on "Interfaces.Stm32.Rcc (spec)"
main.adb:30:06: "System.Bb.Parameters (spec)" depends on "System.Stm32 (spec)"
main.adb:30:06: "System.Stm32 (spec)" depends on "Interfaces.Bit_Types (spec)"
main.adb:30:06: "Main (body)" depends on "Ada.Real_Time (spec)"
main.adb:30:06: "Ada.Real_Time (spec)" depends on "System.Os_Interface (spec)"
main.adb:30:06: "System.Os_Interface (spec)" depends on "System.Multiprocessors (spec)"
main.adb:30:06: "System.Multiprocessors (spec)" depends on "System.Bb.Parameters (spec)"
main.adb:30:06: "System.Bb.Parameters (spec)" depends on "System.Bb.Mcu_Parameters (spec)"
main.adb:30:06: "System.Bb.Mcu_Parameters (spec)" depends on "Interfaces.Stm32.Pwr (spec)"
s-stm32.ads:32:25: "Interfaces" is undefined (more references follow)
gprbuild: *** compilation phase failed
```

I'm using gnat pro 7.4.2 : 

```
lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library$ arm-eabi-gnat --version
GNAT Pro 7.4.2 (20160527-49)
```

I havent' find references to Interfaces.Stm32.Rcc but I find something close : the files *s-stmrcc.ads* defines a package System.STM32F4.RCC that we can expect to implement such interface.

My guess is you're using an more up-to-date toolchain but I dont exclude an configuration problem on my side :)

Fabien-Chouteau, 2016-09-01:
The reference tool chain is GNAT GPL 2016. Note that we use git submodule now, so be sure to clone the repository with --recursive:
`$ git clone --recursive https://github.com/AdaCore/Certyflie.git`

VictorLSogilis, 2016-09-02:
I just tested with GNAT GPL 2016 and I was indeed an toolchain issue. I always assumed Gnat GPL 2016 was a subset of Gnat Pro 2016 so I'm a little surprised but the problem is solved.

closed 2016-09-02 by VictorLSogilis

Fabien-Chouteau, 2016-09-02:
I don't think it's a tool chain problem, it must be the clone --recursive that solved this problem. Ada_Drivers_Library (and therefore Certyflie) uses a new GitHub repo were we started to put run-times for the community [1].

[1] https://github.com/AdaCore/embedded-runtimes

VictorLSogilis, 2016-09-02:
I just tested it by switching the toolchain without touching the git repo and it confirms a toolchain issue : with gnat GPL 2016 it works, with gnat Pro 2016 I have the Interfaces.STM32.Rcc error.

Here are the test with gnat pro, i show you the commits I'm using for Certyflie, the Ada drivers and the embedded toolchains.

```
lambret@debian-sogilis:~/CAP2018/Certyflie$ gprbuild -P cf_ada_spark.gpr -p
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g main.adb
main.adb:30:06: "Interfaces.Stm32.Rcc" is not a predefined library unit
main.adb:30:06: "System.Bb.Parameters (spec)" depends on "Interfaces.Stm32.Rcc (spec)"
main.adb:30:06: "System.Bb.Parameters (spec)" depends on "System.Stm32 (spec)"
main.adb:30:06: "System.Stm32 (spec)" depends on "Interfaces.Bit_Types (spec)"
main.adb:30:06: "Main (body)" depends on "Ada.Real_Time (spec)"
main.adb:30:06: "Ada.Real_Time (spec)" depends on "System.Os_Interface (spec)"
main.adb:30:06: "System.Os_Interface (spec)" depends on "System.Multiprocessors (spec)"
main.adb:30:06: "System.Multiprocessors (spec)" depends on "System.Bb.Parameters (spec)"
main.adb:30:06: "System.Bb.Parameters (spec)" depends on "System.Bb.Mcu_Parameters (spec)"
main.adb:30:06: "System.Bb.Mcu_Parameters (spec)" depends on "Interfaces.Stm32.Pwr (spec)"
s-stm32.ads:32:25: "Interfaces" is undefined (more references follow)
gprbuild: *** compilation phase failed
lambret@debian-sogilis:~/CAP2018/Certyflie$ git status
Sur la branche ravenscar-cf-stable
Votre branche est à jour avec 'origin/ravenscar-cf-stable'.
rien à valider, la copie de travail est propre
lambret@debian-sogilis:~/CAP2018/Certyflie$ cd Ada_Drivers_Library/
lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library$ git status
HEAD détachée sur 34344c2
rien à valider, la copie de travail est propre
lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library$ cd embedded-runtimes/
lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library/embedded-runtimes$ git statusHEAD détachée sur 5cca890
rien à valider, la copie de travail est propre
lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library/embedded-runtimes$ arm-eabi-gnat --version
GNAT Pro 7.4.2 (20160527-49)
Copyright (C) 1996-2016, Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
See your AdaCore support agreement for details of warranty and support.
If you do not have a current support agreement, then there is absolutely
no warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR
PURPOSE.
```

Now I'm testing with the GPL toolchain. Between those two tests I just changed the setting of my bashrc to use the GPL version, that's why I'm starting with a bash command.

```
lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library/embedded-runtimes$ bash
lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library/embedded-runtimes$ arm-eabi-gnat --version
GNAT GPL 2016 (20160515-49)
Copyright (C) 1996-2016, Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

lambret@debian-sogilis:~/CAP2018/Certyflie/Ada_Drivers_Library/embedded-runtimes$ cd ../..
lambret@debian-sogilis:~/CAP2018/Certyflie$ gprbuild -P cf_ada_spark.gpr -p
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g main.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g config.ads
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g crazyflie_system.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g types.ads
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g commander.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g communication.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g imu.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g leds.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g memory.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g motors.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g power_management.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g stabilizer.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g safety.ads
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g crtp.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g console.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g crtp_service.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g link_interface.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g platform_service.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g filter.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g log.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g syslink.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g controller.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g free_fall.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g lps25h.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g pid.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g pid_parameters.ads
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g sensfusion6.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g generic_queue.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g radiolink.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g uart_syslink.adb
arm-eabi-gcc -c -gnatQ -gnatyg -O0 -gnata -gnatwae -g maths.adb
gprbind main.bexch
arm-eabi-gnatbind main.ali
arm-eabi-gcc -c b__main.adb
arm-eabi-gcc main.o -o cflie.elf
```

If you need more data to understand the problem I would be happy to provide them.

Fabien-Chouteau, 2016-09-02:
I checked with @lambourg, there's indeed an incompatibility with GNAT Pro 7.4.2 because of file naming in the run-times. Please ask for a wavefront if you want to continue working with GNAT Pro.