AdaCore/Certyflie #6: Problem with Ada tasking on the Crazyflie

rveenker, 2016-12-12:
There appears to be a problem in the runtime with tasking. This occurs on the Crazyflie after running fine for a couple of minutes. For some reason all tasks simply stop running without any reason (no exceptions are raised either). This causes the Crazyflie to really act to its name and flies like crazy because the main thread that calls the stabilizer_control_loop also stops.

I checked this by adding a small task that does nothing else but toggle a LED. The blinking of this LED eventually stops, but the LEDs for system and battery animations keep going. These LEDs use the Ada.Real_Time.Timing_Events and are not affected. 

To investigate the problem I have turned off (commented out) functionality until I reached a point where the tasks no longer stop (or get blocked). 
When I comment out in uart_syslink.adb:UART_Syslink_Init the call to 'Enable_USART_Rx_Interrupts', the problem seems to have disappeared.

The compiler I use is GNAT GPL 2016 for Linux, running on the bitcraze VM (Ubuntu).

Fabien-Chouteau, 2016-12-13:
Hi @rveenker ,

Thanks for the report! I don't think we ever saw the same behavior. We’ll try to investigate.

Regards,

rveenker, 2016-12-20:
Just an update on my findings regarding this bug.

I realize that, since this bug occurs at random, not every CrazyFlie will behave the same. Some CrazyFlies may not show it at all.
If you have any tests or suggestions on how to debug this problem, please let me know.

For testing I rebuilt the runtime with the debug option (-g).  When I now run the debugger and interrupt the proces after the tasks stopped responding, I end up in s-bbinte.adb:Interrupt_Wrapper at line: Threads.Queues.Change_Priority (Self_Id, Caller_Priority);

The call stack shows:
system.bb.interrupts.interrupt_wrapper
<__gnat_irq_trap>
<signal handler called>
system.bb.board_support.power_down
system.task_primitives.operations.idle
??

Is this the normal place where one would expect the process to halt?

Also I am curious if it is possible to debug (attach) a running CrazyFlie, or start a debug session without flashing the firmware?

Fabien-Chouteau, 2016-12-21:
> Is this the normal place where one would expect the process to halt?

It seems normal, maybe the code is stuck in a interrupt loop, the interrupt is not cleared for some reason.

> Also I am curious if it is possible to debug (attach) a running CrazyFlie,

I think the debugged will always restart the CPU when connecting but I'm not sure...

> or start a debug session without flashing the firmware?

That you can do by just connecting the debugger without using the load command.

rveenker, 2017-01-18:
In searching for a cause of the tasks suddenly stopping, I found a few thing I would like to share.

First I noticed that in stm32-usarts.ads the procedure Disable_DMA_Receive_Requests() sets the value This.Periph.CR3.DMAR to True. I assume this should be False :-)

Next I found that the tasks will keep running when I comment out the line 'Animation.Set_Handler()' in LEDS.LED_Status_Event_Handler: procedure Toggle_LED_Status().
This of course stops the blinking, but perhaps this could be replaced by a task?
Could this have something to do with LED_Status_Event_Handler being a protected object?

Fabien-Chouteau, 2017-01-18:
> In searching for a cause of the tasks suddenly stopping, I found a few thing I would like to share.
> 
> First I noticed that in stm32-usarts.ads the procedure Disable_DMA_Receive_Requests() sets the value This.Periph.CR3.DMAR to True. I assume this should be False :-)

Indeed, thanks for reporting!

> Next I found that the tasks will keep running when I comment out the line 'Animation.Set_Handler()' in LEDS.LED_Status_Event_Handler: procedure Toggle_LED_Status().
> This of course stops the [...]

Thanks for your investigation, it looks like there's a problem with protected objects and interrupt.
How much time does it take to reproduce? Maybe this problem is fixed in the development version of GNAT, I will try to have a look.

Fabien-Chouteau, 2017-01-18:
I opened a pull request for the USART bug: https://github.com/AdaCore/Ada_Drivers_Library/pull/114

rveenker, 2017-01-18:
Not sure if this is slightly hardware-related. My CF sometimes flew fine for minutes.
But on average the tasks would surely have stopped after 5 to 10 minutes.
It also did not make a difference whether I flew the CF or not.

simonjwright, 2017-07-17:
I don’t know whether [this thread on c.l.a.](https://groups.google.com/d/msg/comp.lang.ada/cbCxj-BEYow/NifT4e9ilo4J) is related?

Fabien-Chouteau, 2017-07-17:
Thanks for the link @simonjwright !

simonjwright, 2017-07-26:
I had a _lot_ of trouble with this problem; the board crashed pretty quickly.

I tried upgrading my [cortex-gnat-rts](https://github.com/simonjwright/cortex-gnat-rts) project, based on FreeRTOS, to support this software (see fork [here](https://github.com/simonjwright/Certyflie/blob/sjw)), and the result is that the board is much more reliable.

So I think there just may be a problem in the AdaCore runtime.

simonjwright, 2017-08-01:
I think one possible cause is that the UART has an overrun, but the condition isn’t cleared.

Before making the change below, the board was getting stuck after a few minutes; since, it’s been running (on the desk, not connected to the controller) for 2 hours.

```
diff --git a/drivers/uart_syslink.adb b/drivers/uart_syslink.adb
index fbf349f..447f7cf 100644
--- a/drivers/uart_syslink.adb
+++ b/drivers/uart_syslink.adb
@@ -325,6 +325,18 @@ package body UART_Syslink is
             Clear_Status (NRF_USART, Read_Data_Register_Not_Empty);
             Enqueue (Rx_Queue, Received_Byte);
             Byte_Avalaible := True;
+         elsif Status (NRF_USART, Overrun_Error_Indicated) then
+            --  RM0090 rev11 top of p 1001: overrun is cleared by
+            --  reading SR (which we've already done (twice?!)), then
+            --  reading DR.
+            --
+            --  Assuming the read DR is OK and the incoming character
+            --  is discarded.
+            Received_Byte := T_Uint8 (Current_Input (NRF_USART) and 16#FF#);
+            Clear_Status (NRF_USART, Read_Data_Register_Not_Empty);
+            Clear_Status (NRF_USART, Overrun_Error_Indicated); -- ?
+            Enqueue (Rx_Queue, Received_Byte);
+            Byte_Avalaible := True;
          end if;
       end IRQ_Handler;
 ```

Fabien-Chouteau, 2017-08-01:
Thanks @simonjwright ! I like that better than a bug in the run-time.

Do you mind opening a pull request?

Fabien-Chouteau, 2017-08-03:
Hi @rveenker I just megred Simon's patch. Can you rebuild and try to reproduce the error with the new code?

rveenker, 2017-09-03:
I have built the latest version and no longer see this issue.
Thanks!

Fabien-Chouteau, 2017-09-13:
Great! Thank you @simonjwright  for the patch!

closed 2017-09-13 by Fabien-Chouteau