AdaCore/Certyflie #9: gprbuild error undefined attribute "runtime"

dwwaddell, 2017-05-23:
pi@raspberrypi:~/crazyflie/Certyflie $ gprbuild -P cf_ada_spark.gpr -p
cf_ada_spark.gpr:29:08: undefined attribute "runtime"
crazyflie_full.gpr:10:08: undefined attribute "runtime"
gprbuild: "cf_ada_spark.gpr" processing failed

Fabien-Chouteau, 2017-05-29:
Hello @dwwaddell,

Are you using a cross compiler for ARM? You can download it here: [libre.adacore.com](http://libre.adacore.com/download/configurations).

Regards,

dwwaddell, 2017-05-29:
I am running on a Raspberry Pi 3 and I tried three cross compilers:

1.       

2.      

3.      

 

All three of these platforms provide binaries that won’t run on the Pi:

 

pi@raspberrypi:~/Downloads $ l

total 1274680

drwxr-xr-x  5 pi pi      4096 May 26 13:47 .

drwxr-xr-x 45 pi pi      4096 May 19 20:52 ..

-rw-r--r--  1 pi pi 483502080 May 26 13:44 AdaCore-Download-2017-05-26_2037.tar

drwxr-xr-x  9 pi pi      4096 May 23 11:25 gnat-gpl-2016-arm-elf-linux-bin

-rw-r--r--  1 pi pi 386353827 May 23 10:54 gnat-gpl-2016-arm-elf-linux-bin.tar.gz

drwxr-xr-x  9 pi pi      4096 May 16  2016 gnat-gpl-2016-raspberrypi-linux-linux-bin

-rw-r--r--  1 pi pi 340104304 May 23 14:15 gnat-gpl-2016-raspberrypi-linux-linux-bin.tar.gz

-rw-r--r--  1 pi pi  95257233 May 26 13:37 spark-gpl-2016-x86_64-linux-bin.tar.gz

drwxr-xr-x  3 pi pi      4096 May 26 13:47 x86_64-linux

 

pi@raspberrypi:~/Downloads $ file /home/pi/Downloads/gnat-gpl-2016-arm-elf-linux-bin/bin/gprbuild 

/home/pi/Downloads/gnat-gpl-2016-arm-elf-linux-bin/bin/gprbuild: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.9, stripped

pi@raspberrypi:~/Downloads $ /home/pi/Downloads/gnat-gpl-2016-arm-elf-linux-bin/bin/gprbuild

bash: /home/pi/Downloads/gnat-gpl-2016-arm-elf-linux-bin/bin/gprbuild: cannot execute binary file: Exec format error

 

pi@raspberrypi:~/Downloads $ file gnat-gpl-2016-raspberrypi-linux-linux-bin/bin/gprbuild 

gnat-gpl-2016-raspberrypi-linux-linux-bin/bin/gprbuild: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.9, stripped

pi@raspberrypi:~/Downloads $ gnat-gpl-2016-raspberrypi-linux-linux-bin/bin/gprbuild

bash: gnat-gpl-2016-raspberrypi-linux-linux-bin/bin/gprbuild: cannot execute binary file: Exec format error

 

pi@raspberrypi:~/Downloads $ file /home/pi/Downloads/x86_64-linux/adagpl-2016/gprbuild/gprbuild-gpl-2016-x86_64-linux-bin/bin/gprbuild 

/home/pi/Downloads/x86_64-linux/adagpl-2016/gprbuild/gprbuild-gpl-2016-x86_64-linux-bin/bin/gprbuild: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.9, stripped

pi@raspberrypi:~/Downloads $ /home/pi/Downloads/x86_64-linux/adagpl-2016/gprbuild/gprbuild-gpl-2016-x86_64-linux-bin/bin/gprbuild

bash: /home/pi/Downloads/x86_64-linux/adagpl-2016/gprbuild/gprbuild-gpl-2016-x86_64-linux-bin/bin/gprbuild: cannot execute binary file: Exec format error

The last one is, of course, expected. All three had been installed in /usr/gnat with doinstall. The only executable I could get to run was the one installed with “sudo apt-get install gnat”:

 

pi@raspberrypi:~/Downloads $ which gprbuild

/usr/bin/gprbuild

pi@raspberrypi:~/Downloads $ file gprbuild

/usr/bin/gprbuild: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=9fe6d7d7ba0f3848110da3e3e324178b62768c12, stripped

Any help is appreciated.

Dave

dwwaddell, 2017-05-29:
Well, it took a lot of reading but I finally figured out there is no cross-compiler for the Crazyflie CPU (STM32F4) on the Raspberry Pi 3.

 

 <http://campusebikes.com/> Campus eBikes

2777 Woodland Pk Dr #1213

Houston, TX 77082

+1-281-529-5640

Fabien-Chouteau, 2017-05-30:
That's right, we have binary distribution of the cross compiler for Windows and Linux x86/x86_64 hosts.

Regards,

closed 2017-05-30 by Fabien-Chouteau