AdaCore/Certyflie #10: Building Issue:  a-stream.ads:65:09: expected type "Standard.Boolean

hponso, 2017-06-26:
Hi, I downloaded the last version of Certyflie and tied to build it using GNAT GPL 2017 ARM ELF with the recommended line " gprbuild -P cf_ada_spark.gpr -p ". 
I'm getting this errors:
`
...
main.adb:54:11: (style) incorrect line terminator
main.adb:55:10: (style) incorrect line terminator
a-stream.ads:65:09: expected type "Standard.Boolean"
a-stream.ads:65:09: found type "Ada.Tags.Size_Ptr"
a-stream.ads:65:09: expected type "Ada.Tags.Select_Specific_Data_Ptr"
a-stream.ads:65:09: found a composite type
a-stream.ads:65:09: no value supplied for component "Tags_Table"
gprbuild: *** compilation phase failed
[2017-06-26 11:06:21] process exited with status 4, 0% (1/530), elapsed time: 00.78s 

´

Can someone see what I'm missing?

I did not installed SPARK Discovery from AdaCore  if that helps.

Thanks for any help!!.

Fabien-Chouteau, 2017-06-26:
Hi,

We did not ported this demo to the new GNAT GPL 2017 release yet. We recommend to use the 2016 release that is still available at [http://libre.adacore.com/download/configurations](http://libre.adacore.com/download/configurations.).

Regards,

closed 2017-06-26 by Fabien-Chouteau

hponso, 2017-06-26:
Thank you so much!!