AdaCore/Certyflie #11: Can’t build with GNAT GPL 2017

simonjwright, 2017-06-26:
With both a Linux download and a macOS rebuild, we get (modulo line numbers)
```
$ gprbuild -p
using project file cf_ada_spark.gpr
Compile
   [Ada]          main.adb
a-stream.ads:65:09: expected type "Standard.Boolean"
a-stream.ads:65:09: found type "Ada.Tags.Size_Ptr"
a-stream.ads:65:09: expected type "Ada.Tags.Select_Specific_Data_Ptr"
a-stream.ads:65:09: found a composite type
a-stream.ads:65:09: no value supplied for component "Tags_Table"
gprbuild: *** compilation phase failed 
```
Same replacing the `embedded-runtimes` submodule with the latest commit.
Just building the RTS:
```
lockheed:ravenscar-stm32f407disco simon$ gprbuild -P ravenscar_build.gpr -XRTS=ravenscar-full
Compile
   [Asm_Cpp]      start-rom.s
   [Asm_Cpp]      start-ram.s
   [Asm_Cpp]      start-common.s
   [Asm_Cpp]      handler.s
   [C]            raise-gcc.c
/Users/simon/adacore/embedded-runtimes/base_runtimes/ravenscar-full/common/raise-gcc.c: In function 'exception_class_eq':
/Users/simon/adacore/embedded-runtimes/base_runtimes/ravenscar-full/common/raise-gcc.c:869:10: warning: implicit declaration of function 'memcmp' [-Wimplicit-function-declaration]
   return memcmp (except->common.exception_class, ec, 8) == 0;
          ^~~~~~
   [C]            adaint-xi.c
   [Ada]          s-lisisq.adb
   [Ada]          s-lidosq.adb
  ...
   [Ada]          s-string.adb
   [Ada]          s-strhas.adb
   [Ada]          s-stratt.adb
a-stream.ads:65:09: expected type "Standard.Boolean"
a-stream.ads:65:09: found type "Ada.Tags.Size_Ptr"
a-stream.ads:65:09: expected type "Ada.Tags.Select_Specific_Data_Ptr"
a-stream.ads:65:09: found a composite type
a-stream.ads:65:09: no value supplied for component "Tags_Table"
gprbuild: *** compilation phase failed
```

Fabien-Chouteau, 2017-06-26:
Hi Simon,

I just answered to #10 on the same subject. We did not ported this demo to the new GNAT GPL 2017 release yet. We recommend to use the 2016 release that is still available at http://libre.adacore.com/download/configurations.

Regards,

closed 2017-06-26 by Fabien-Chouteau

Fabien-Chouteau, 2017-06-30:
Hi @simonjwright,

We've just updated this demo for GNAT GPL 2017. Give it a try and don't hesitate to tell us if it doesn't work.

Regards,

simonjwright, 2017-06-30:
Hi Fabien,

Great stuff! we have a compilation! (some googling about how to deal with recursive submodule pull; and nervous deletion of `embedded-runtimes`).

_Must_ order my Crazyflie directly.

By the way, this was an excellent test of my arm-eabi build of GNAT GPL 2017 on macOS, will be uploaded to Sourceforge Real Soon Now.

simonjwright, 2017-07-27:
@Fabien-Chouteau, this is a bug in GNAT GPL 2017, isn’t it? The output with `-gnatG` shows that the internal structures are quite mangled. Should I report it via report@adacore.com?

It’d help to know whether there’s any workround ...

Fabien-Chouteau, 2017-07-27:
Sorry @simonjwright, I don't understand this question. Is it related to the issue reported above?

simonjwright, 2017-07-27:
@Fabien-Chouteau, the compiler should have been able to compile that code;  there’s some weird interaction between the location of the file and the runtime that confuses the compiler. If AdaCore already know about it, no need for me to report it again (unless of course the plan is to replicate this into the FSF compiler). If you’re not sure, I’ll report it anyway.

simonjwright, 2017-07-27:
@Fabien-Chouteau - Found the problem: in GPL 2017 the private component `Ada.Tags.Type_Specific_Data.Type_Is_Abstract` changed its name to `Is_Abstract`. This is, I expect, because the component takes its name from the public accessor function, which should have been called `Is_Abstract` but was previously called `Type_Is_Abstract`. The change was made in the FSF GCC sources (i.e. GGG 8) 2017-04-25.