AdaCore/Certyflie #12: Executable loads over CF bootloader

simonjwright, 2017-07-11:
The [Crazyflie memory map](https://wiki.bitcraze.io/doc:crazyflie:bootloader:protocol) has the bootloader at `0x08000000` and the main firmware at `0x08004000`, whereas Certyflie loads at `0x08000000`, thus overwriting the bootloader and preventing over-the-air updates.
I’m not sure how this could be easily corrected; the load address is fixed at low level in the runtime, so at first glance we’d need a new runtime. Perhaps the loader script could be parameterised.

AnthonyLeonardoGracio, 2017-07-12:
Hi Simon,

Indeed, we already know this issue. I will discuss with my colleagues to see if we can provide a custom linker script with a load address set at 0x08004000 instead of using the one defined in the used runtime. We will keep you posted.

Regards,

simonjwright, 2017-07-12:
I’ve done a bit of googling, and this works for me (it’s for an Arduino Due):

In the linker script,
```
START_OFFSET = DEFINED (START_OFFSET) ? START_OFFSET : 0;

MEMORY
{
    flash (RX)  : ORIGIN = 0x00080000 + START_OFFSET,
                  LENGTH = 512K - START_OFFSET
    sram (RWX)  : ORIGIN = 0x20070000, LENGTH = 96K
}
```
and then build with
```
gprbuild -P testbed -largs -Wl,--defsym,START_OFFSET=0x1000
```
(similarly to the way you already allow the stack size to be overridden).

Fabien-Chouteau, 2017-07-17:
Hi @simonjwright ,

There might be a problem with vector table address. I don't know how the bootloader works but the run-time was not tested in this configuration so there's a chance it won't be compatible.

Did you try yourself Simon?

simonjwright, 2017-07-17:
It seemed to work! (aside from running away, as I said on issue #6).

`start-rom.S` says that `VTOR` is set in `handler.S`, but in fact it’s set in `s-bbcppr.adb`:
```
      VTOR := System_Vectors'Address;
```
so all should be OK.

simonjwright, 2017-07-26:
Works for me: see [linker script](https://github.com/simonjwright/cortex-gnat-rts/blob/master/stm32f4/adalib/stm32f407-flash.ld) and corresponding [GPR](https://github.com/simonjwright/Certyflie/blob/sjw/crazyflie.gpr).

(These are from my project [cortex-gnat-rts](https://github.com/simonjwright/cortex-gnat-rts) and my [fork](https://github.com/simonjwright/Certyflie/blob/sjw) of this project to use that RTS, motivated by issue #6 here)