AdaCore/Certyflie #13: gprbuild error

pierreciv, 2017-07-13:
Hello everyone,

I have an error when I try to build the firmware : 

:~/Documents/GT/Certyflie$ gprbuild -P cf_ada_spark.gpr -p
cf_ada_spark.gpr:26:08: undefined attribute "create_missing_dirs"
cf_ada_spark.gpr:30:08: undefined attribute "runtime"
crazyflie_full.gpr:16:08: undefined attribute "runtime"
crazyflie_full.gpr:18:26: unknown attribute "runtime"
config.gpr:19:09: variable "target" is not typed
config.gpr:20:12: illegal case label "riscv32-unknown-elf"
stm32f40x.gpr:11:08: undefined attribute "create_missing_dirs"
HAL.gpr:11:08: undefined attribute "create_missing_dirs"
cortex_m4f.gpr:15:08: undefined attribute "create_missing_dirs"
components.gpr:14:08: undefined attribute "create_missing_dirs"
middleware.gpr:18:08: undefined attribute "create_missing_dirs"
ravenscar_support.gpr:13:08: undefined attribute "create_missing_dirs"
gprbuild: "cf_ada_spark.gpr" processing failed

I guess, I have correctly installed the compiler for ARM (I went on http://libre.adacore.com/download/configurations, selected both the platform "ARM ELF Format" and the platform "x86-64 GNU Linux (64 bits)", downloaded all the linked files and then installed the GNAT GPL 2017 for the 2 platforms, following the given instructions)

Here is what is in my file :

:~/Documents/GT/Certyflie$ ls
Ada_Drivers_Library  cf_ada_spark.gpr  config  drivers  gnat.adc  hal  init  lib  LICENSE.txt  modules  README.md  types  utils

Have you an idea ?

Thank you

simonjwright, 2017-07-17:
Are you sure you’re running the correct `gprbuild`? (possible `PATH` problem).

Fabien-Chouteau, 2017-07-17:
Hello @pierreciv ,

> middleware.gpr:18:08: undefined attribute "create_missing_dirs"

This attribute was added to GNAT Pro 17 and GNAT GPL 2017, so you are probably not using the correct version of `gprbuild`.

You can check the version with:
`$  gprbuild --version`

Regards,

@simonjwright thanks for helping :+1:

closed 2017-07-17 by Fabien-Chouteau